jim-p/Change_Filament #1: Expose functionality via API

gdombiak, 2019-05-31:
Hey Jim,

I love this plugin. If you could expose the functionality of this plugin via an API then I can use your plugin and its API from [OctoPod](https://github.com/gdombiak/OctoPod). OctoPod is free and open source with thousands of users that could use your plugin from the app. :)

Thanks,
Gaston

jim-p, 2019-05-31:
I can look into this, but it might be a while. I'm just dipping my toes into the water with OctoPrint plugins so I don't know all the details yet. I see a some examples in the plugin development docs, though, so it might not be too difficult to do. I don't have any personal apple devices around so I can't do much with OctoPod directly to test, though.

OllisGit, 2019-10-01:
@gdombiak @jim-p ,
IMHO this is not possible with the current implementation, because the complete logic is executed in JavaScript.
To provide an API you must transfer the logic to the server (python-code).

A simple REST-API looks like this: https://github.com/OllisGit/OctoPrint-KnowledgeBase/wiki/SendDataToServer#simple-rest-api

tazran, 2021-05-09:
This might help: https://docs.octoprint.org/en/master/modules/plugin.html?highlight=endpoint#octoprint.plugin.types.BlueprintPlugin