jim-p/Change_Filament #2: Feature Request and/or Question: Multimaterial Support

Patronics, 2019-06-01:
Does this plugin support multimaterial printers  such as Prusa's MMU2? This plugin would be invaluable if it has or could get support for multimaterial printers

jim-p, 2019-06-01:
Not yet, but it looks like it wouldn't be too difficult to have a config knob to set the number of tools, or maybe pull it from the printer profile, and print additional lines of unload/load buttons.

Patronics, 2019-06-01:
That sound easy enough for printers with multiple extruders, but note that with for example Prusa's MMU2 where multiple filament lines are sharing one extruder, I suspect the logic for controlling that would be more complicated than you describe. Though it'd definitely be a good start to implement a mode for multiple standalone extruders, where your description should work fine.

jim-p, 2019-06-01:
Yeah that would be a much larger challenge, but if I can find the gcode to select the other extruders it may not be that much more complicated. With multiple extruders it's just a matter of using T0/T1/etc to pick the right tool from what I can see.

Of course if someone wants to give me a Prusa w/MMU2 I'd be more than happy to figure it all out ;-)
/s