jim-p/Change_Filament #4: Visualization doesn't respect CSS

rlogiacco, 2019-06-03:
I believe you are using non-standard fonts in your plugin as shown by the attached screenshot where buttons labels are displayed with a different font and there are missing icons
[image]

jim-p, 2019-06-03:
Those are Font Awesome icons, and the OctoPrint UI loads the font for those, so I'd consider them standard. I'm not setting anything to control the fonts otherwise: https://github.com/jim-p/Change_Filament/blob/master/octoprint_Change_Filament/static/js/Change_Filament.js#L43

Perhaps your browser is preventing those from being loaded? The icons from that font are used all over the OctoPrint UI.

rlogiacco, 2019-06-03:
I understand your point, but as you can see from the attached screenshot everything else on the page (including awesome icons) is loaded properly..

[image]

For what I can see inspecting the HTML my best guess is the issue is related to the fact you are setting font awesome on the whole button instead of using the `i` tag with the `fa fa-something` classes, but I only had a quick look at it...

jim-p, 2019-06-04:
That is a valid way of using font awesome icons for buttons. Perhaps whatever is altering your CSS did not do a complete job? I know it can also be done using `<i>`, `<span>` and so on, but it works fine with the default theme.

rlogiacco, 2019-06-04:
I can try to revert to the default theme just to double check, anyway I'm using the Themeify plugin, so nothing particularly customized

jim-p, 2019-06-04:
You might raise the issue with them as well. It may be that nothing else had used that format yet and they didn't account for it in their plugin.

rlogiacco, 2019-06-04:
I have run a test and yes, I confirm that, when the themeify plugin is disabled, the buttons are partially displayed correctly (see attached screenshot where the text font and buttons size still don't match up with the rest of the page).

[image]

To my humble opinion, it is your plugin fragment requiring a fix: anywhere else in the Octoprint web interface the font awesome icons are either used with `i` or `span`, while you are using it on a `button` item. 

I could submit a pull request, but I'm pretty sure you don't need that: the change is so simple you would have done it already if you were agreeing with my point of view.

But being an open source project, other than providing my personal feedback, I can only say _"thank you for donating it to us 👍 "_

jim-p, 2019-06-04:
I know I could change it, and I probably will, but I still contend that it needs fixed in the Themify plugin if they are not supporting CSS supported by OctoPrint itself. It needs brought up to them either way.

jim-p, 2019-08-04:
I tried a few different ways to make this work and I'm not seeing a way to make it happen using `i` or `span`. Mostly because using HTML in the `name` parameter does not work. It isn't rendered as HTML, so the tags show through. I didn't see a way to form the tag using the customControls parameters yet either. If anyone has some pointers to another example that works, I'd love to see it so I can adapt what I've got.

At the moment the best fix is still to bring it up with Themify so they can fix their plugin to support this style.

tablatronix, 2020-01-01:
Can someone link to the themify issue, this is really annoying.