jim-p/Change_Filament #5: [Q] while pringting

boelle, 2019-06-06:
is there any way i could use this while printing without M600 enabled?

i mean press pause in octoprint, then park and chanage filament and then press pause again to continue the print?

jim-p, 2019-06-06:
I doubt it would work out since the nozzle would not return to the correct position afterward. Might vary from printer to printer with that behavior, though. You could pause and then raise Z manually then unload/load and then lower Z the same amount, then try to resume. I don't have a lot of confidence in that working but if you try it out, let me know how it goes.

boelle, 2019-06-07:
i ended up with just pausing and then changing the filament as fast i could

it kind of works, if the nozzle stops inside infill you will not see it, but if it does at an perimeter it will show but not much, the faster the change is the better

but my thought was that since octoprint is in control this plugin could do the pause, then park, and user can unload and load and then "unpause" and return to where it can from and continue the print

Fabrizio3, 2019-07-14:
dont know about your firmware, but with klipper im obligated to home X and Y before continuing the print, so i pause, home those 2, change my filament and continue flawlessly give it a try!
also yeah, an all in one button would be really nice, might have to get into programming and make something like that, just use the octoprint pause, raise Z, home XY, unload, load and continue

closed 2019-08-04 by jim-p