jim-p/Change_Filament #6: Settings / parameters

Fabrizio3, 2019-07-14:
the load and speed settings dont update when clicking save in octoprint's interface, i have to restart the octoprint instance (wich is a hassle).
there are some settings not available to the user that would be really really helpful like the retraction when parking and such (i dont know exactly how it does it, maybe your plugin is not the fault) but i want 0 retraction when loading
also maybe a way to incorporate the whole sequence for us with no m600 comand? just by issuing the regular octoprint pause, Z lift, X and Y home, unload and wait for the load button to be pressed (will look into it myself to see if i can contribute to you with that)
but all around i have to say you've made a great plugin! its exactly what ive been wishing for, great work!

Fabrizio3, 2019-07-14:
ill try to tinker with the code as it doesnt seem to dificult... at least i hope so
things i want:
-adding a purge (with user set parameters?)
-changing the retract (maybe also user accesible?)
-sanity check (unload total distance is the same as load total distance including retracts and purges so you dont mess up the extrusion of the print because its absolute not relative, meaning it aboslutely needs to be the same)

Fabrizio3, 2019-07-14:
also adding a blocking command when starting the macro, because accidental double clicking can cause a BIG mess, just a sanity check (mainly cause my mouse is trash)

jim-p, 2019-07-15:
> the load and speed settings dont update when clicking save in octoprint's interface, i have to restart the octoprint instance (wich is a hassle).

That's a limitation of pulling the settings the way the plugin has to. You don't need to restart the interface, just refresh the browser. I'm not sure there is a good way around that.

> there are some settings not available to the user that would be really really helpful like the retraction when parking and such (i dont know exactly how it does it, maybe your plugin is not the fault) but i want 0 retraction when loading

I was just thinking about that as well. I put it in there because that's what the built-in filament swap in Marlin does and I was copying its logic. You generally do want a smidge of retraction before a move like that if you are mid-print, because otherwise it'll leave a string/blob as it drags away. Though the way I tend to use the plugin after a month or so now, I don't like it on either. I could make a checkbox/toggle to turn it on/off.

> also maybe a way to incorporate the whole sequence for us with no m600 comand? just by issuing the regular octoprint pause, Z lift, X and Y home, unload and wait for the load button to be pressed (will look into it myself to see if i can contribute to you with that)

Might be possible for that, but when I was messing around with formatting, adding more buttons made it a bit more awkward, especially if I split them into multiple lines. If I can find a way to do that and not make it look bad, I'll see what I can do.

> -adding a purge (with user set parameters?)

That's what load does, unless I'm not understanding what you're after there.

> -sanity check (unload total distance is the same as load total distance including retracts and purges so you dont mess up the extrusion of the print because its absolute not relative, meaning it aboslutely needs to be the same. like if the print move is E206.54 and your commands are not right, when the next move ensues like E206.93 if you're not at 206.54 youll have problems, either a retract or a extrude you dont want).

The plugin issues M83 to do a relative extrude in either direction so that shouldn't really matter in the long run, I thought. The built-in M600 swap unloads way more than it loads, so I don't know that it would be different here. The presumption is that unload retracts the entire length of the bowden tube, but load extrudes only enough to purge the nozzle.

> also adding a blocking command when starting the macro, because accidental double clicking can cause a BIG mess, just a sanity check (mainly cause my mouse is trash)

That's a good idea, too. Though knowing when to re-enable it is a bit trickier.

The code is very straightforward in most cases for these things. I know there are a few things on my list I need to fix, open issues here and some other problems like I didn't fix the M117 text for load so it's not putting the right text on the display. I just haven't had a chance to get back to updating this. Hopefully will very soon!

Fabrizio3, 2019-07-15:
> That's what load does, unless I'm not understanding what you're after there.

i have a bowden system, i load 545mm of filament at 3000mm/m and that goes right to the nozzle with about 1mm to spare, but then after that it would be a QOL thing to have it purge a set ammount automatically
>That's a good idea, too. Though knowing when to re-enable it is a bit trickier.

octoprint issues a command that says "busy" and doesnt allow more commands to be used until the action is finished, i think it does it when homing or something, but blocking for 1s could be just right
>Might be possible for that, but when I was messing around with formatting, adding more buttons made it a bit more awkward, especially if I split them into multiple lines. If I can find a way to do that and not make it look bad, I'll see what I can do.

on that note octoprint has a command "M0" that issues the octoprint pause, wich is very useful
you could implement that somewhere (ill try to test it out for you first then comment about it)
also i personally have to issue a home so i could implement that, i have to learn how the code works and i may be able to do some stuff like checkboxes and extra settings
> The plugin issues M83 to do a relative extrude in either direction so that shouldn't really matter in the long run, I thought. The built-in M600 swap unloads way more than it loads, so I don't know [...]

but if you say, are at E200, retract 60 (so youre at E140) and load 70 (E210) and the gcode says E201 because thats the next line, wouldnt it cause trouble? or is there a way to remember the last E position before the sequence and issue a G92 E (last position) and have it continue as if nothing happened?

jim-p, 2019-08-04:
Some of the issues raised here should be fixed in the latest version, but not 100% yet.