jim-p/Change_Filament #7: Work with Ender 3?

scubieman, 2019-07-18:
I tried using this with ender 3. The park button works but the unload and load button does not.

I did try preheating the nozzle.

jim-p, 2019-07-18:
It works on my CR-10S so I don't see why it wouldn't work there. It's all standard Marlin stuff.

If you switch to the terminal tab after clicking the button(s), do you see any errors there? You may have to suppress temperature messages on that tab to see anything useful. Maybe open the UI up in two browser windows so one can watch the console while the other clicks the button.

scubieman, 2019-07-19:
Send: M117 Unloading filament
Send: M83
Recv: ok
Send: G1 E-500 F1600
Recv: ok
Send: M18 E
Recv: echo: too long extrusion prevented
Recv: ok
Send: M117 Replace filament, set new temp, click Load
Recv: ok
Recv: ok

jim-p, 2019-07-19:
I'm not sure what your firmware considers "too long" for an extrusion. The default unload length is `500` and the default load length is `50`. If neither button works, then even `50` might be too long.

In the settings for this plugin, try using a smaller unload and load length. Start at `25` or so and work your way up.

You could also try an alternate firmware like TH3D UF to remove that unnecessarily low limit.

I don't see that this could be a bug in the plugin, so I'll close this for now, but please do follow up if you find some settings that work for your printer.

closed 2019-07-19 by jim-p

scubieman, 2019-07-19:
Jim. I know closed.

 Found that this one only allows 200 MM of extrude/extract. but 400 MM of filament is in the tube. So I need to run twice. Just a FYI.