jim-p/Change_Filament #8: Park and Unload boxes blank

telboy595, 2019-07-23:
Hi i have installed your plug in but the 2 boxes Park and Unload are blank with no name or color to them. I have not tried to change filament eyt as i have only now managd to find out what these 2 buttons do.

Hope you can help resolve the small problem.

I am using a Anycubic i3 mega with Octoprint

Terry

jim-p, 2019-07-23:
What other plugins do you have installed? Anything that might change the theme/style of OctoPrint?

telboy595, 2019-07-23:
Hi Jim

Many thanks for your prompt reply to a newbie to Octoprint.

I have the following plugs in installed :-

Action Command Prompt
Action Trigger
Change Filament
Enclosure Plug in
Filament manager
Navbar temperature
Preheat Button
Printtime genius
Tab order
Themeifly

Hope it helps

Best regards

Terry Warner

KB Agency
Tel          (44) 1604 404862
Mobile   (44) 7899 668079

DISCLAIMER:
The information in this E-Mail message is confidential and may be legally privileged. It is intended solely for the addressee and access to this message by anyone else is unauthorised. If you have received the message in error, please notify us immediately and delete it. If you are not the intended recipient, any disclosure, copying, distribution or any other use is prohibited and is be unlawful.

jim-p, 2019-07-23:
This almost certainly a duplicate of #4 then -- There is a known issue with Themify.

closed 2019-07-23 by jim-p