jim-p/Change_Filament #9: Hotend too low when returning to print

ScottyDS, 2019-08-11:
Perhaps more of a question rather than an issue.
After clicking the M600 button (Themify has broken the other buttons so I can only see theM600), and changing the filament, the z height is too low by about 2 steps. (When the head returns to the print, I quickly grab the z couplers and click them twice, and the print carries on normally.) If I dont do this, the nozzle crashes into the job, and makes a terrible mess of things.

I have inserted an M600 command in the Gcode test file to see if it is my Marlin settings causing the issue, but the filament change executes flawlessly.

As soon as I use the M600 plugin button, the problem happened again........However, After restarting the Printer, and the Octoprint server (complete power off and on again), all seems to be working as expected.

Has anyone prehaps had a similar issue?  Is it something I could have done wrong?

Now that it is working, it is bloody marvelous. Helped me out of 2 fillament breakages.

jim-p, 2019-08-11:
I haven't seen that happen here, but I don't use the M600 button much. The actions taken by M600 are 100% in the firmware of the printer, though. Neither this plugin nor OctoPrint controls what it does beyond issuing the Marlin command to the printer.

closed 2019-08-11 by jim-p

ScottyDS, 2019-08-12:
Thanks Jim.

I have since tested it many more times and it works perfectly
 Thanks for sharing your work.
Regards
Duane.