jim-p/Change_Filament #10: Feature Request:  lock steppers and extrude a line when purging

rlsmith46, 2019-08-18:
It would be very helpful upon homing to:
1..Extend stepper motor timeout to a long time (say 10 minutes)
2.Turn the stepper motors on so that the position is not changed when filament is loaded.

Upon resuming print it would be nice it a long line was printed a few MM from the left edge of the bed as the purge cycle.
Thanks for you plugin and consideration of my request.

jim-p, 2019-08-18:
1. Any particular reason to do that vs enabling without a disable timer? (`M84 S300` vs `M84 S0`)?
2. At least for my printers it seems they are still enabled after the park move, and no command was sent to disable them. Sending `M17` should be harmless, but I didn't think it was necessary.
3. Some people use this to swap filament mid-print, so going back to the bed after for a wipe would be harmful. After loading the filament there isn't really a need for a wipe at that point. Either you are mid-print and you can just pick the excess off with some tweezers or you are between prints and your print start gcode can do the wipe.

rlsmith46, 2019-08-19:
thanks for the quick reply.
Answer to 1 & 2
I have and Ender 3 Pro.. After executing the park move, the  steppers are
turned off.  ( I have done the equivalent operation with a Cura script with
the same effect.. . the steppers turn off after the park.  I also want to
change filament mid- print.  In googling around to try to get a resolution,
I have determined that this is a common problem.  The issuance of a M17
alone doesn't resolve the issue, since on my Ender the default time to turn
off the steppers with no additional commands seems to be 60 seconds,,,
which is not sufficient time to change the filaments out.

So my hope would be to incorporate something like an :

M18 S1800
M17

to allow the steppers to remain on for 30 minutes followed by a  turn on
all steppers

Answer to 3

I too want to use the plugin to change filament and resume print.

I don't know enough about Marlin & g code to know if if would be possible
to store an absolute position value ( rather than relative) to restore the
head to before resuming printing.

If that were a possibility, my recommendation to print a purge line rather
than just extrude makes sense to me.
If not, my recommendation is in fact senseless.

4. New request.... since you have already done the heavy lifting  in your
plugin, it would be icing on the cake to  have  user- selectable  variables
to set the pre-unload and pre-load temperatures..
Thanks again for a wonderful plugin and for listening to my ranblings

This will enable me to change filament without a layer shift caused by me
inadvertently moving the z axis.

I'm not sure if you would want to issue a M18 with a paramater to turn off
only the Extruder stepper or not.

jim-p, 2019-08-19:
1/2 should be fine, I'll try to add that next time I'm working on it, perhaps with a configurable timeout value.

3. Storing the position in a plugin like this is not feasible. There are ways to do it, but the way this one works it's pretty much write-only from the code to the printer. Besides that, trying to move the head back down to the bed with a print on the bed is dangerous and a recipe for disaster. There is no benefit to doing a wipe on the bed after a mid-air filament swap that I can see. After it purges, just wait a few moments and remove the excess filament by hand in some way.

4. I've thought about doing old and new filament temp values but that would change so often it doesn't seem like a great idea to do it here the same way as other settings. I'll keep that in mind for the future, though, maybe I'll come up with a viable method.

tablatronix, 2019-12-22:
Can the extruder motor be excluded, so we can use manual knob for loading assist etc..

rlsmith46, 2019-12-22:
yes. The request to lock the steppers is particularly important because I
presumed a manual load.
Without the steppers being locked, the manual load is likely to cause a
layer shift.