jim-p/Change_Filament #11: Feature Request: Extrude a little bit, before unload

OllisGit, 2019-09-26:
First: Thx, for the plugin!

Small improvement will help me a lot:
Please add a feature to extrude filament (e.g.slowly 3-5mm or adjustable) before the retract command is send. 
The reason (in my case) is that the filament is not "under pressure" and the retraction is not working, because the filament is not hot enough at the nose.

Thx in advance
Olli

jim-p, 2019-09-26:
It should be easy enough to add configuration option to set a pre-unload extrude. I thought about the same thing, I'll try to work it in next time I'm working on the plugin.

JoeScotty, 2022-07-21:
Hi, did you habe a chance to Insert this Feature?
I'm allways extruding some filament before unloading to have less hot filament inside and avoid strings and cloggs. Would be Nixe to add this with parameters length and speed. Thanks!