jim-p/Change_Filament #12: Feature request

broncosis, 2019-10-20:
Why not have a option to do the preheat for the user 
I have a gcode file I use for this 

it preheats extrudes a tiny bit then retracts the filament pauses so that I get a notification from octoslack and then I load the new filament hit resume and it extrudes a little to prime and then shuts down

jim-p, 2019-10-21:
It could, but with everyone using different filaments/hotends/etc it would add a lot of complexity for little benefit. You can already preheat on the control tab in two clicks. If I add it into this package, then I need to add settings for preheat temperatures (for old and new material), which also means that you'd have to change that and refresh any time you change material, etc.

So maybe in the future, but so far I haven't felt that it would be convenient enough to warrant the work.

broncosis, 2019-10-23:
That's a valid point. 

⁣Get BlueMail for Android ​

sepodele, 2020-06-09:
I'd very much like this as well. Lots of us are using only one type of filament and would need just one config for temperature. Please reconsider. :)

ussdeveloper, 2020-12-02:
I agree a lot of cases is related with for example switch PLA to PLA ... plugin could add two more inputs for pre-heat temp and checkbox enable/disable pre-heat feature. It would be handful to not wait for head gets proper temp to click unload and then cool down by finger .... please reconsider. :)

mdziekon, 2023-06-16:
For anyone still interested, I've just added such option in my fork of this plugin here:
https://github.com/mdziekon/Octoprint-ChangeFilament

It's limited to M600 button only, and currently allows to select only from available temp profiles, but it's a start.