jim-p/Change_Filament #15: Handling EXTRUDE_MAXLENGTH

tablatronix, 2019-12-22:
I am assuming this plugin is not working for me because my `EXTRUDE_MAXLENGTH` is 200mm

Can you detect the error Ill have to check it comes through terminal exactly what it says.

In settings allow extrude settings to be multiple moves, so 200mm x 2

tablatronix, 2019-12-22:
refs

`#define EXTRUDE_MAXLENGTH (X_MAX_LENGTH+Y_MAX_LENGTH) //prevent extrusion of very large distances.`

`too long extrusion prevented`

tablatronix, 2019-12-22:
it seems `M302` can disable something, looking into if it does both cold and lengthy flags in marlin.
I was thinking it disables `PREVENT_DANGEROUS_EXTRUDE` which covers both.

jim-p, 2019-12-22:
The way this plugin is made, there isn't a way for it to read or interpret values returned from the printer. In your special case, you could set the unload/load length to half what you really want, and simply click the button twice.

If it becomes a more widespread issue I could maybe work around it some other way, like perhaps supporting comma-separated values in the fields and running multiple commands. For example, setting the unload length to `150,150` would result in a total of 300 issued as two commands.

tablatronix, 2019-12-22:
Good idea

jerryfudd, 2020-03-17:
Hi, I was coming onto report the same issue and had the same idea of being able to set the extrude lengths - however could it be possible to also set a multiplier in the plugin i.e. extrude "100" multiplied by "5" and it just repeats to give you "500" full unload - likewise for load?
Possibly also adding in a box to define the pause duration between repeating extrude commands as you can see that being the next issue.

JHale716, 2020-05-04:
(Posting for content on this thread from #21)
The maximum I have managed to have my Ender 3 Pro unload is 100mm, and the full filament unload length is 475mm (slightly shorter Bowden tube fitted than the factory 500mm)

Is it possible to have the plugin feature filament unload length and maximum machine retract length, and then manage the retract from there with multiple commands?

tablatronix, 2020-05-04:
But then you are asking the user to fill out 2 inputs, seems cumbersome, it would be cool if you could query marlin for `EXTRUDE_MAXLENGTH` but you cannot afaik

JHale716, 2020-05-04:
Yes, two inputs, which are entered once and set and forget after that...

sepodele, 2020-09-01:
I second this. 👍

furyhh, 2021-10-01:
I'm new to this stuff, I just started editing marlin for my 3D printer, can someone explain to me what EXTRUDE MAXLENGTH means and how to set it up