jim-p/Change_Filament #17: Park Button configuration suggestion

cstaab, 2020-01-19:
I really like the plugin. but I do have a suggestion that I think would make the plugin even more usable.  Could you provide a user configurable option (check box) to allow Z axis to be either a relative change or an absolute position?

jim-p, 2020-01-19:
That should be possible, my main concern would be someone setting that too low or not realizing what it will do, and then end up crashing the print head or gantry into the object being printed.

But I could see that being useful, especially on small printers, to always go to a position near the max of Z.

cstaab, 2020-01-19:
I see your concern.  Make the default relative and add a warning about the
risks of using absolute.

Thanks for considering the suggestion.