jim-p/Change_Filament #18: Compatibility with Python3 and Octoprint 1.4.0

ASMfreaK, 2020-02-16:
I've tried installing this plugin on Octoprint 1.4.0rc3.
The plugin installed without any errors, but the frontend didn't change a bit.

jim-p, 2020-03-04:
Probably because I need to add the bits to mark it as compatible. I meant to test this locally in a test setup but now that 1.4.0 is out I'll test it on one of my live printers. Should be fairly easy once I manage to get my installs updated to 1.4.0.

https://github.com/foosel/OctoPrint/releases/tag/1.4.0
https://octoprint.org/blog/2020/03/04/new-release-candidate-1.4.0/

closed 2020-03-05 by jim-p

jim-p, 2020-03-05:
Upstream change also needed: https://github.com/OctoPrint/plugins.octoprint.org/pull/408

Once that is merged it should start showing up.

jim-p, 2020-03-05:
Upstream change merged, plugin shows up for installation on OctoPrint 1.4.0 with Python 3.x, installs and shows up as expected.

ASMfreaK, 2020-03-08:
Thanks! It works great!

commit 1eb29ff9 (2020-03-05): OctoPrint 1.4.0/Python 3.x compatibility. Fixes #18