jim-p/Change_Filament #21: Ender 3 pro unload does not work

elmarfaber, 2020-04-24:
Hello,

the unload function does not work for me under ender 3 pro.
Park and load works fine

jim-p, 2020-04-24:
Most likely the same issue as #15 -- the unload value is larger than your firmware allows. Set a lower unload length and maybe click the button 2x as a test.

closed 2020-04-24 by jim-p

JHale716, 2020-05-04:
The maximum I have managed to have my Ender 3 Pro unload is 100mm, and the full filament unload length is 475mm (slightly shorter Bowden tube fitted than the factory 500mm)

Is it possible to have the plugin feature filament unload length and maximum machine retract length, and then manage the retract from there with multiple commands?

jim-p, 2020-05-04:
Not yet -- That's what #15 is about adding