jim-p/Change_Filament #25: Display issue when using Themify and Material UI light theme

JeffB42, 2020-07-03:
I have two different installations of OctoPrint, both with the Themify and Change Filament plugins installed, and this issue occurs with both, so I thought I would post this issue.  I have the Themify plugin installed, and have the theme set to Material UI light.  The background colors of the Park and Unload buttons are white, and with the font's color also being white, it makes the buttons unreadable:

[image]

If I change the theme to a dark theme, like discorded, I'm able to read the text:

[image]

...but unfortunately, I don't use any of the dark themes, I use the only light theme provided, Material UI light.

I've noticed that there are other inconsistencies with other themes as well, like when using Disoranged, the M600 button has a background color of red, but the other buttons have a dark gray background color:

[image]

jim-p, 2020-07-03:
Those are issues in Themify and/or those themes. This package specifies how the buttons and icons should look using standard bootstrap classes, which the Themify package and/or themes are free to redefine.

closed 2020-07-03 by jim-p

JeffB42, 2020-07-04:
Hi Jim, I'm a developer.  It's been a while since I've used Bootstrap, so I don't recall "what's standard" or what "the right way to do it" is (wrt Bootstrap), but looking at the other buttons within OctoPrint (and this includes the base OctoPrint as well as the other plugins that I have installed), every other case of where there's a button, the markup is along the lines of:

`<button class="btn btn-primary">
	<i class="fa fa-check"></i>
	<span data-bind="text: 'Park')">
		Park
	</span>
</button>`

Where as, the markup in Change Filament is just a button element:
`<button class="btn nowrap btn-info fa fa-check">
     Park
</button>`
(additional markup was removed and omited for clarity)

Here's what's displayed when I changed the second button to the markup I listed above:
[image]

Notice now that the font has also been fixed as well.

I don't know if this issue is a bug in Themify, or in Change Filament, but it seems that everywhere else in OctoPrint (including other plugins) the markup of the buttons is along the lines of <button><i></i><span></span></button>, instead of <button></button>

jim-p, 2020-07-04:
It could also be a fault or missing feature in OctoPrint's plugin system. This plugin defines the buttons using the ViewModel and not via HTML directly. So it is not something that can be changed by me, since I don't control that output. 

See this for example: https://github.com/jim-p/Change_Filament/blob/master/octoprint_Change_Filament/static/js/Change_Filament.js#L92

There is no way I have yet found using that system to define the parameters in a way which results in the "good" HTML.

But using classes in the way this plugin does works everywhere else, and is valid for bootstrap and font awesome.